enesbcs/mpyeasy-async #1: Can't set static IP

Budman1758, 2022-09-23:
Just got this setup on a Wiznet W5100s module. Trying to set a static IP and after setting and clicking submit the page reverts to the original settings. I also unchecked the dhcp box.  In the serial console I see this error message every time I try this.

"Loading settings
conf invalid syntax for integer with base 10"

When I go to the settings page the ip address shown in the setting is not the address the module actually has either. See screenshot.
[image]

I uncheck dhcp and change to what I want, hit submit and page reverts to same as before.

enesbcs, 2022-11-28:
Saving IP fixed in commit https://github.com/enesbcs/mpyeasy-async/commit/e90610b2690b4a0af9025bf4c15391c0365d89ef

Warning: Please uncheck DHCP if you want to use fixed IP

Budman1758, 2023-03-27:
Just got back to testing this and still cannot set static IP. DHCP is disabled. Using latest release from the releases page. PI Wifi version.

enesbcs, 2023-04-01:
Sorry @Budman1758 but i am unable to reproduce your issue.
I have tested it with mPyEasy-async 1.1.332 with Pico W module.

I've got 192.168.2.11 IP address first from DHCP.
Then i went to config menu:
- unchecked DHCP
- changed IP, Mask, GW and DNS line as all four are required.

 [image]

Pressed submit then rebooted the unit.
And PICO W has not available as 192.168.2.12

--------------------
In case you want to get debug console you need to connect with USB, start Thonny,set interpreter as RP2040 and check "Interrupt working program on connect"

Now a Shell console is available with >>>
press CTRL-D here to restart the device and the logs should tell if something gone wrong.

enesbcs, 2023-04-01:
After few more tries now i managed to reproduce it, but it only happens when Thonny is not connected.. when Thonny is connected fix ip works. Which makes debugging a bit hard. :-/

enesbcs, 2023-04-01:
@Budman1758 could you try with this binary?
https://github.com/enesbcs/mpyeasy-async/releases/download/v1.1.332-beta/picow_firmware.uf2

Altered a little the second ifconfig() in the source it now works for me without connecting with Thonny

Budman1758, 2023-04-01:
That seems to have fixed it. Worked the first try.

enesbcs, 2023-04-01:
Thanks for testing, interesting that only the Wifi model suffered from this bug.

closed 2023-04-01 by enesbcs