enesbcs/mpyeasy-async #3: NTP not working

Budman1758, 2022-09-23:
Not sure if it's supposed to be but NTP does not seem to be working.  I have a timeserver IP address defined in the setting.

enesbcs, 2022-11-06:
yep, ntptime library is not contained in RP2040 port only at the ESP port... but i am testing now with custom manifest file.

enesbcs, 2022-11-28:
ntptime lib freezed in RP2 in v1.1.332
https://github.com/enesbcs/mpyeasy-async/releases/tag/v1.1.332-beta

Budman1758, 2023-04-01:
Using the binary from the [static IP issue](https://github.com/enesbcs/mpyeasy-async/releases/download/v1.1.332-beta/picow_firmware.uf2) still no NTP working.  Have it enabled and tried NTP server name set blank, with hostname and with IP address set and not working at all. Tried rebooting every time I changed host also. Tried several host names and IP addresses.

enesbcs, 2023-04-02:
Sorry, i tried with and without Thonny, my PICO W modul has always got the correct time.
Tried with DNS name 'hu.pool.ntp.org' and my own local NTP server with IP address also. Try to debug it with Thonny console.

Budman1758, 2023-04-02:
Seems that some of the addresses I found and tried either don't work or MPYEasy does not like them. Tried  "pool.ntp.org"  and suddenly it works.  Tried several from [this page](https://tf.nist.gov/tf-cgi/servers.cgi) and none of the ones I tried worked.  Does MPYEasy use any particular "hard coded" address? Does it need an address in the settings? On ESPEasy I never put an address for NTP.  Always blank.  

Guess this can be closed.

enesbcs, 2023-04-02:
Micropython ntptime unit has a default host: "pool.ntp.org"
Usually i am using the hungarian pool "hu.pool.ntp.org", except in my own network, where i have an own NTP server on my OpenWRT router for my all devices inside.

closed 2023-04-02 by enesbcs