enesbcs/mpyeasy-async #4: Unit number always reverts back to "0"

Budman1758, 2022-09-24:
I have ESPEasy P2P controller in use and that seems to work fine. After a reboot the unit number I assign always reverts back to zero.

enesbcs, 2022-09-25:
Thanks for testing it has to be a settings loading error.

Budman1758, 2022-09-26:
Another thing I notice is on the main page where the node list is it shows all other nodes that I have but the unit itself does not show in it like it does in RPIEasy or ESPEasy units. Probably not a big deal but thought I would mention it.

enesbcs, 2022-11-28:
Yes mpyeasy and mpyeasy-async sends broadcast UDP messages to each node, but it seems, that it never gets back its own packages, so nodes did not show on its own listing... although it is not impossible to add some extra code to show itself on the node list.

enesbcs, 2022-11-28:
Unit number saving fixed in commit https://github.com/enesbcs/mpyeasy-async/commit/e90610b2690b4a0af9025bf4c15391c0365d89ef

Budman1758, 2023-04-01:
Saving unit number is working now. Don't know if you added the code to show itself on the main page or not but that does not work yet. If not worried about that (I'm not) then feel free to close this one.

closed 2023-04-02 by enesbcs