enesbcs/mpyeasy-async #5: RPI Pico WIFI module seems to do nothing

Budman1758, 2022-09-28:
I dunno if its me screwing up or what. Have 3 brand new pico w modules. Have "dropped" the file into all of them and absolutely nothing happens. After the file uploads to the module the drive disappears from my file manager and thats it. I assume the module is rebooting but I never get a WIFI AP to connect to so I can configure it and absolutely nothing when connected to the serial port. No output in the console or any reaction to keyboard input.

Ideas?  Pretty sure I'm using the correct file.    https://github.com/enesbcs/mpyeasy-async/releases/download/v1.1.206-beta/mpyeasy-11260-pico-w.uf2

TIA

enesbcs, 2022-09-28:
Did you use Thonny to see the console? Do you see any files on the unit, after pressing ctr-c on the console?

Budman1758, 2022-09-28:
Not using Thonny. Was using putty to see the serial output. That worked on the other module I'm testing. Will install Thonny and see what's up.

Budman1758, 2022-09-28:
I get this in the Thonny window after a soft reboot.

MPY: soft reboot
Loading settings
Loading AdvSettings
Init hardware...
Loading HW settings
Loading Pinout settings
hwic spic
Init error, failing: local variable referenced before assignment
Starting main loop
Main loop ended

enesbcs, 2022-09-29:
It looks like failed to create and save the dataset, will try to reset my own PicoW module and load uf2 again to try to reproduce this.

enesbcs, 2022-11-28:
Changed init to try network connection at startup forever in v1.1.332

closed 2023-04-01 by enesbcs