enesbcs/mpyeasy-async #7: GPIO command only responds to "1"

Budman1758, 2022-09-30:
Wiznet 5100 module.  Hardware page set D0 to output.  Reboot.  Tools page command line.  "GPIO,0,1" turns led on. Command echos in the log.  "GPIO,0,0" does nothing. LED stays on. No output in the log. No device for this in device page.

LED draws less than 4 milliamps.

Device page. Add generic switch. Select gpio 0. Hit submit. Device becomes unresponsive. Web page "refused connection".

At this point the only way I can reconnect to the device is to complete wipe and reload the firmware. All settings are lost then.

In Thonny I almost always get unable to connect to com port. Permission denied.  Nothing else is using the com port.

My hair gets thinner by the day....    :>)

enesbcs, 2022-10-01:
Weird, as the only function i have tested was the GPIO before rolled out first alpha test build and it worked.

Did you use multiple tabs in the browser to the device? It may influence saving configs, as only one buffer fits in memory to serve webpages, and may crash.

Budman1758, 2022-10-01:
I have had more than 1 tab open, yes. Not sure if that's the problem as I have a couple open right now and doing multiple things and haven't crashed it yet. Will try again making sure only 1 tab open.

Budman1758, 2022-10-01:
Well.... Crashed again...

enesbcs, 2022-10-01:
>>File "mpyeasy.py", line 237, in networkInit
>>KeyboardInterrupt:
>>MicroPython v1.19.1-209-g0c45a28d2-dirty on 2022-07-29; W5100S-EVB-Pico with RP2040

The only information i see from that is the python application interrupted at the LAN init stage. It may occur If it is forced to fix ip mode and the parameters are malformed, or setted to dhcp mode and dhcp response never arrive, or used gpio 16-21 which are reserved by the lan module.

Budman1758, 2022-10-01:
Fresh install. No other settings touched. Device page. Device 1 "Input - Switch Device/Generic GPIO", Set to push button active low. Reboot. Hardware page. Set GPIO 0 to "Input pullup". Reboot. Module crash. Repeatable and happens every time.

If I reverse the steps above same thing happens. Below is what shows in Thonny. B4 the module crash there are zero problems with lan connectivity.  Is there any other way to "See" whats happening on my end?

Traceback (most recent call last):
  File "boot.py", line 1, in <module>
  File "mpyeasy.py", line 488, in <module>
  File "uasyncio/core.py", line 1, in run_forever
  File "uasyncio/core.py", line 1, in run_until_complete
  File "mpyeasy.py", line 426, in main
  File "mpyeasy.py", line 234, in networkInit
  File "inc/rp2/unet.py", line 56, in lan_init

Wiznet module.

enesbcs, 2022-11-06:
Seems that queued commands stucked on Tools page.. fixing in progress

enesbcs, 2022-11-28:
Commands fixed in commit https://github.com/enesbcs/mpyeasy-async/commit/e90610b2690b4a0af9025bf4c15391c0365d89ef

closed 2024-02-01 by enesbcs