enesbcs/mpyeasy-async #12: Pico 2 compatibility?

Budman1758, 2024-09-09:
Hello.  Long time no mess with....lol 
Just picked up a couple Pico 2 boards and tried to install the firmware to one.  Firmware copies to the "drive" when plugged in but it never seems to create a visible WIFI AP to connect to.

enesbcs, 2024-09-10:
I think its not, as Pico 2 has no wifi, so i think it will not connect to an AP. 
Otherwise PicoW and Pico2 has different chip onboard, so i almost sure, that the current binary which compiled for the Pico1 makes nothing more than a segfault error.

Budman1758, 2024-09-10:
Well....Why the heck was I thinking it had WIFI?   lol.  I bought a PI Zero 2 W at the same time and guess I just thought they had it.  I tell ya, it's hell getting old.....

enesbcs, 2024-09-11:
Although i guess a Pico 2W will be released in the near future. :)

Budman1758, 2024-12-04:
> Although i guess a Pico 2W will be released in the near future. :)

Got a couple on the way....  Stay tuned.  :)

Budman1758, 2024-12-08:
Got my Pico 2's.  This time with Wi-Fi.  :)

Copied the firmware file to the Pico and after that I get nothing at all.   No AP or any indication of activity at all.  Thonny reads nothing from the module, and I get the drive in Windows whenever I plug it in regardless of whether the boot button is pressed when plugging in or not.  In other words, the device is in mass storage mode all the time its plugged into the USB port.  No AP when plugged into a power only port either.  

Please let me know where to go from here.

tonhuisman, 2024-12-08:
Not sure if the Pico and Pico 2 are binairy compatible 🤔

Budman1758, 2024-12-08:
> Not sure if the Pico and Pico 2 are binairy compatible 🤔

That sounds ominous...... 😨

tonhuisman, 2024-12-08:
> Not sure if the Pico and Pico 2 are binairy compatible 🤔

They aren't backward nor forward compatible: https://forums.raspberrypi.com/viewtopic.php?t=375262

If you have a microPython version for the Pico 2 you can however run existing python scripts from the Pico 1, as the IO hardware is compatible.
You can recompile your binaries for the RP2350 (Pico 2) platform, and upload that to the unit.

NB: I don't own any of the RPi Pico devices, so it's all from a little googling I did and reading the release info for the Pico 2 (and knowledge I gathered over the years). No guarantees... 🤷‍♂️

Budman1758, 2024-12-08:
I was hoping this software would work on it.  I don't really have the skills to do that.  Thats why I like the "ESPEasy" way of doing this stuff.  It's "easy"  😃 .

Maybe the author can recompile it to work.....  (hint hint nudge nudge)  😉

Budman1758, 2025-04-08:
Just curious if there is any progress on Pico 2 compatibility?

enesbcs, 2025-04-09:
As i see the Pico 2 W is now available in the market. But honestly I haven't written a single line of code since I moved and changed job.  I don't have much free time lately.

enesbcs, 2025-04-11:
Unfortunately micropython semantics changes inbetween different versions, so compilation of code which is written for an earlier micropython release can be challanging... last time developers changed a parameter in wifi connection it caused me an hour headache to debug. :D

Budman1758, 2025-04-11:
Well, hopefully you can find some time to see what can be done.  Thanks for all you do.  :)